joan2937/lg #9: unexpected error on kernel 5.11

amo93-2020, 2021-08-04:
Hi,
i am trying lgpio in python on ubuntu 21.04 with kernel 5.11 but i get 'unexpected error' when calling 'gpio_claim_output ()' function.
with kernel 5.8 lgpio it works

joan2937, 2021-08-04:
@amo93-2020 

Thanks for the issue.  I have had a quick look and there does appear to be a problem.

I am not sure when I'll have time to look into this properly.  It may take a day or two.

joan2937, 2021-08-05:
I have looked a little bit more.  The /dev/gpiochip API has been broken by the introduction of v2 of the API.  To say the least I am not impressed.

It will probably take a day or so to judge the ramifications of the various API breaks.

amo93-2020, 2021-08-06:
thanks for the support!, hope you can update this soon

joan2937, 2021-08-10:
Just an update.  It'll be several more days at least before I prepare a new release, probably by Sunday at the latest.

stephanrotolante, 2021-08-12:
@amo93-2020 Is this the error you are getting?

```
Traceback (most recent call last):
  File "/home/ubuntu/blink_test.py", line 14, in <module>
    lgpio.gpio_claim_output(h, LED)
  File "/usr/lib/python3/dist-packages/lgpio.py", line 773, in gpio_claim_output
    return _u2i(_lgpio._gpio_claim_output(handle&0xffff, lFlags, gpio, level))
  File "/usr/lib/python3/dist-packages/lgpio.py", line 461, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'

```

@joan2937 If you need help making updates I'd me more than happy to help 😄  - Although keep  in mind  I come from a blissful world of nodejs, web development. I. have never created programs like this so there will be some learning curve initially

joan2937, 2021-08-12:
@stephanrotolante 

Thanks for the offer.

The problem at the moment isn't the code changes which need to be made, it's more to do with deciding if I want to deprecate parts of the existing lg API  to accommodate the changes to the /dev/gpiochip API or not.

I think I've come to the conclusion that I will incorporate the changes without breaking the existing lg API.

Given the way the kernel people have implemented the /dev/gpiochip API I can see there will be more breakages downstream (I can see /dev/gpiochip API2 being broken by /dev/gpiochip API3).

amo93-2020, 2021-08-24:
Perfect!
Now the library also works with the 5.11 kernel.
thank's for your job

closed 2021-08-24 by amo93-2020

warthog618, 2024-04-23:
> I have looked a little bit more. The /dev/gpiochip API has been broken by the introduction of v2 of the API. To say the least I am not impressed.

Can you provide some detail as to what you found that was broken, as the uAPI v2 should not have altered the uAPI v1 at all.  That was certainly the intent and the change did not impact libgpiod nor any of the other libraries using the uAPI that I am aware of.
IIRC, there may've been some tightening up of the unused sections of the structs, which must now be zeroed, but that was unrelated to the introduction of uAPI v2.

I'm only running across this ticket now by accident, and it would've been good if you could've dropped us a line at the time, but I would still like to understand your problem.  To say the least...