joan2937/lg #10: multiple i2c_open for same port/address returns validate handles

PSkeberdis, 2021-08-05:
Expecting that the first i2c_open for s specific port/address would provide a valid handle, but that a second i2c_open for the same port/address without a close would not provide a valid handle.

First instance of program run well, then second instance stomps the first instance as it also receives a valid handle.

Am testing on Pi4B+ running Ubuntu 21.04:
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	        21.04
Codename:	hirsute

Code snippet from lcd_i2c.py:
...
import lgpio
import time
from RPLCD.i2c import CharLCD

from datetime import datetime

address = 0x27

cols = 20
rows = 4

lcd = CharLCD(i2c_expander='PCF8574', address=address, port=1,
              cols=cols, rows=rows,
              charmap='A02',				# A00, A02 or ST0B
              backlight_enabled=True)
...

Fix would be to only issue a valid handle for I2C Port/address if a current handle for the same I2C Port/Address did not exist.

joan2937, 2021-08-05:
That is by design.  If the underlying operating system is happy to have the same device opened multiple times then so is lg.

PSkeberdis, 2021-08-08:
I don’t agree with your design philosophy in that regard. The device driver is part of the OS and should be responsible for deconflicting device access.

 

Here’s a reference for Linux device drivers from https://www.linuxjournal.com/article/2476:

 

Do I Really Need to Write a Driver?

There are several reasons for writing our own device driver:

*         To solve concurrency problems when two or more processes try to access a device at the same time.

*         To use hardware interrupts: as the kernel runs in protected mode, the user cannot manage interrupts directly from a program.

*         To handle other unusual applications, such as managing a virtual device (a RAM disk or a device simulator).

*         To obtain satisfaction as a programmer: writing a driver increases personal motivation as well as control over the computer.

*         To learn about the internal parts of the system.

closed 2021-08-15 by joan2937