joan2937/lg #14: Disable Hardware Chip Select

savejeff, 2022-08-20:
Hi!

Absolutely love the library. Took me only half a day to implement Arduino style wrapper for digital write, read, i2c and spi. 

With SPI i ran into a problem: You can not control the SPI CS manually. the problem is with code like :
```
	SPI.beginTransaction();

	//Read Operation: (0)(Addr)
	SPI.transfer(((reg & 0x7F) | 0x00));
	uint8_t val = _jSPI->transfer(0x00);
	
	SPI.endTransaction();
```

Here the CS is pulled up and down again between the transfer calls. The Arduino behavior is that CS is low between beginTransaction and endTransaction. 

My dirty fix solution is for now to use an extra GPIO and avoid usage of the dedicated hardware CS Pin. Is there by chance to disable the SPI driver from controlling the CS pin via the flags or something like that? That would be very nice

Otherwise very well designed library!
Thx

joan2937, 2022-08-29:
Sorry for the delay.

lg is intended to be a pure Linux library in that it will only support features provided by Linux.

I have not looked at the SPI driver recently.  Do you know if it supports such a usage?

Otherwise you would probably have to just not connect the "Linux" SPI CS and assign and connect a spare GPIO to toggle yourself.

savejeff, 2022-09-03:
No Problem.

I was just checking if there might be a convenient solution. 

i found some documentation and it seems like disabled the cs is not possible. i solved it as you described it where i use an additional free gpio as software cs and just avoid using the spi hardware cs pin

closed 2022-09-03 by savejeff