joan2937/lg #15: SPI Pin Mode on Init

savejeff, 2022-09-25:
On the Raspberry PI, I have found that SPI stops working, meaning the pins like SCK and MOSI stop working when they are set to input or output manually at any time after boot. 

this means when for example the SCK pin is set as input at the end of a program after ending spi conversation, the next time the program is executed spi does not work anymore. 

might there be an explicit pin mux configuration call missing at the lgSpiOpen function or is there any other way to configure the spi pins back to their default function after using them as digital input or output?

joan2937, 2022-09-25:
The claim functions will only succeed if the GPIO are not assigned.  That means the GPIO have not been assigned to SPI.

The new `/dev/gpiochip` interface (which lg uses) provides no mechanism to set the pin mode.  That is another pretty fundamental failure of the way `/dev/gpiochip` works.

On the Pi you will have to use something like `pigpio` to set the GPIO to SPI mode.  There is no generic way of setting the GPIO mode in Linux.

savejeff, 2022-09-26:
Okay i understand. That's unfortunate but can be avoided by just not using the pins as digital input or output. 
I'll look into pigpio to switch the gpio. 

But I'm again and again in awe of how little functionality is standardized in 2022. I'm working on a cross-platform (Windows, Linux, Mac) project and there is only since C++17 standardized access to the file system and still no standard c++ support for sockets, the most basic way to communicate over network.

closed 2022-09-26 by savejeff