joan2937/lg #17: Unable to install lgpio python module

adamhl8, 2023-02-03:
I'm not very familiar with python in general, so apologies if I'm missing something obvious here. Ultimately I'm trying to figure why I can't import lgpio.

```
ModuleNotFoundError: No module named 'lgpio'
```

I followed the instructions on the site:

```
sudo apt install swig python3-dev python3-setuptools
wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make
sudo make install
```

Both `make` and `sudo make install` run without issue. Here's the output from the latter:

```
install -m 0644 lgpio.h                  /usr/local/include
install -m 0644 rgpio.h                  /usr/local/include
install -m 0755 liblgpio.so.1 /usr/local/lib
install -m 0755 librgpio.so.1 /usr/local/lib
install -m 0755 rgpiod                   /usr/local/bin
install -m 0755 rgs                      /usr/local/bin
install -m 0644 rgpiod.1                 /usr/local/man/man1
install -m 0644 rgs.1                    /usr/local/man/man1
install -m 0644 lgpio.3                  /usr/local/man/man3
install -m 0644 rgpio.3                  /usr/local/man/man3
ldconfig
/usr/bin/python3
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:146: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/bin/python3
```

In the `PY_LGPIO` directory I tried running `sudo python3 setup.py install` but I get the following error(s) (omitted the beginning of the output):

```
running install
...
...
file lgpio.py (for module lgpio) not found
file lgpio.py (for module lgpio) not found
running build_ext
building '_lgpio' extension
creating build
creating build/temp.linux-aarch64-cpython-311
aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c lgpio_wrap.c -o build/temp.linux-aarch64-cpython-311/lgpio_wrap.o
cc1: fatal error: lgpio_wrap.c: No such file or directory
compilation terminated.
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
```

I'm not really sure what else to try. Any help is greatly appreciated.

joan2937, 2023-02-03:
`sudo python3 setup.py install` is not part of the build instructions.

Try again in a clean directory.

```
wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make
sudo make install
```

Are you using Python2 (deprecated) or Python3 or both?

Start Python at the command line

e.g. python or python2 or python3 and import lgpio

What happens?

adamhl8, 2023-02-03:
Using python3. python2 is not installed. I'm on Debian sid (`Linux pi 6.1.0-3-arm64 #1 SMP Debian 6.1.8-1 (2023-01-29) aarch64 GNU/Linux`).

I tried again in a clean directory.

```
~/lg$ python3
Python 3.11.1 (main, Dec 31 2022, 10:23:59) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import lgpio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'lgpio'
```

joan2937, 2023-02-03:
Try both the following again.

```
sudo apt install swig python-dev python3-dev
sudo apt install python-setuptools python3-setuptools
```

adamhl8, 2023-02-03:
`python-dev` and `python-setuptools` are not available on sid. Not sure if those packages are required for proper installation when I have the python3 versions.

I'm able to install `python-dev-is-python3` as a replacement for `python-dev`. I tried installing again in the same (clean) way as above but I'm still not able to import.

joan2937, 2023-02-03:
Perhaps sid is the problem.  If so I can't help as I stick with stable.

I have the following Python packages manually installed.

```
python-dev-is-python2/stable,stable,now 2.7.18-9 all [installed]
python-setuptools/stable,stable,now 44.1.1-1 all [installed]
python3-dev/stable,now 3.9.2-3 arm64 [installed]
python3-gpiozero/stable,stable,now 1.6.2-1 all [installed]
python3-pip/stable,stable,now 20.3.4-4+rpt1+deb11u1 all [installed]
python3-setuptools/stable,stable,now 52.0.0-4 all [installed]
```

```
Linux ceres 6.1.0-v8+ #1609 SMP PREEMPT Fri Dec 16 15:11:24 GMT 2022 aarch64 GNU/Linux
```

```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

adamhl8, 2023-02-03:
As a workaround, I'm able to install the latest `python3-lgpio` package from [launchpad](https://launchpad.net/ubuntu/lunar/arm64/python3-lgpio/0.2.0.0-0ubuntu2) (python3-lgpio_0.2.0.0-0ubuntu2_arm64.deb). I can import lgpio without issue after installing that package (and dependencies). Seems like my issue has something to do with those missing python packages.

Thank you for your time, I appreciate the help.

closed 2023-02-03 by adamhl8