joan2937/lg #20: Enumerate valid chips

EliRibble, 2023-05-31:
It appears that the only way to determine how many chips are valid is to attempt `gpiochip_open` with some number of different integers and discard any failures. Is there an API to get the list of chips numbers that are valid? If not, what is the range of integers we should test?

joan2937, 2023-05-31:
The Linux kernel will assign gpiochip numbers.  I assume they would be assigned in order 0, 1, 2, etc.  I can think of no reason for not assigning in that order.  SO try opening increasing numbers until there is a failure.

However that is my assumption.  I do not know if the kernel makes any guarantee.

EliRibble, 2023-05-31:
Thanks - is there a way to distinguish between "that chip does not exist" and "the chip exists, but you can't get a handle to it"?

DanielDecker, 2024-04-08:
Type the following command in a the terminal to list all available gpiochips: `ls /dev/gpiochip*`
For more details try: `sudo cat /sys/kernel/debug/gpio`
I hope this helps!