joan2937/lg #24: Installing into a Python virtual environment?

MatthewGerber, 2024-01-04:
@joan2937 Do you have an example (or any previous experience) installing this into a Python virtual environment? I tried various modifications of your makefile, but I haven't been successful. Since virtual environments are a standard way of installing packages, it would seem like a nice enhancement of what you have here.

joan2937, 2024-01-05:
I'm afraid my knowledge of Python is limited and my knowledge of Python virtual environments (venvs) is non-existent.

Have a look through https://forums.raspberrypi.com/viewtopic.php?t=358841

waveform80, 2024-03-29:
There should now be wheels available for the [lgpio](https://pypi.org/project/lgpio/#files) and [rgpio](https://pypi.org/project/rgpio/#files) projects on PyPI that should work on the armhf architecture. I'm still working on aarch64 builds as PyPI refuses to accept wheels for aarch64 unless they're tagged "manylinux". Unfortunately building those is proving trickier than expected because all the pre-baked containers have linux headers that are too old, so I'm going to need to bake my own build environment for that.

waveform80, 2024-04-01:
Well, that took a bit more effort than expected but there should now be statically linked wheels for armv7l and aarch64 on PyPI for CPython versions 3.9 thru 3.12.

apr-d, 2024-04-09:
Hi Joan,

Can Lgpio function on Ubuntu 22.04, or is it exclusively compatible with 21.04? Thank you!

Kind regards,
Luke

waveform80, 2024-04-09:
It should function happily on ubuntu 22.04, either installed as a system package (`sudo apt install python3-lgpio`), or (since #25) pip installed in a venv (`pip install lgpio`).

Incidentally, if you do find a problem with lg on ubuntu, please feel free to file a bug under the [lg-gpio package](https://bugs.launchpad.net/ubuntu/+source/lg-gpio/+filebug) (unfortunately I wasn't allowed to just call it "lg"). If it's a packaging issue, that's my fault, and I can deal with it there without bugging upstream here! :)

pdgilbert, 2025-12-29:
Thanks @CamDavidsonPilon . Filtering out some of your extra stuff, it seems what I needed was `sudo apt install liblgpio-dev`. I'll summarize once I do a bit more testing.