joan2937/lg #30: Can gpio in lgGpioReport_t be changed to uint16_t?

vickash, 2024-07-30:
I'm using an Orange Pi Zero 2W. For whatever reason, the GPIO numbers on this exceed 255, so when using `lgGpioSetSamplesFunc`, I get reports where the GPIO numbers overflow. Reports from GPIO 258 appear to be coming from GPIO 2 for example.

joan2937, 2024-07-31:
I will have to find some time to look into this.  I thought that gpiochip GPIO numbers were constarined to be 0-63.

lg is a very low priority for me - don't expect a response soon.

vickash, 2024-08-12:
I understand. I forked, changed it to uint16, and it works fine so far. I'll use that in the meantime. Opened a [PR](#31) for the change as well.

warthog618, 2024-08-16:
GPIO numbers (i.e. offsets) are NOT _constrained_ to 0-63.  You are probably thinking of the number of lines in a request which is limited to 64, cos who would want to request more than 64 lines at once, but individual GPIO chips can and do exceed that so offsets can be larger.
The GPIO uAPI defines offsets as a __u32, so you should go with that.

vickash, 2024-08-16:
Updated my PR to 32 bits.